discos/discos #515: Management container crashes without custom logger

jjpopski, 2020-02-11:
If scheduler componet fails loading custom logger, an unknown exception isn't catched and the Management Container crashes down.
To reproduce the bug, start Management container without activating logging container; then open the operator input. At this point, the management container crashes.

aorlati, 2020-02-12:
This issue awkwardly looks like a issue we has some time ago that should be already fixed.

aorlati, 2020-02-13:
My investigations show that the seg-fault happens in ACS code. I guess when releasing the component after an exception is thrown by the execute method.

jjpopski, 2020-02-13:
#0  0x00007ffff6671495 in raise (sig=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:64
#1  0x00007ffff6672c75 in abort () at abort.c:92
#2  0x00007ffff6f29beb in __cxxabiv1::__terminate (
    handler=<value optimized out>)
    at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:41
#3  0x00007ffff6f29354 in __cxxabiv1::__cxa_call_unexpected (
    exc_obj_in=<value optimized out>)
    at ../../../../libstdc++-v3/libsupc++/eh_personality.cc:781
#4  0x00007fffda94ce04 in CCore::execute (this=<value optimized out>)
    at Core.cpp:32
#5  0x00007fffda8eefbd in SchedulerImpl::execute (this=0x7fffb4003bd0)
    at SchedulerImpl.cpp:79
#6  0x00007ffff4198456 in acscomponent::ACSComponentImpl::__execute (
    this=0x7fffb4003bd0) at acscomponentImpl.cpp:171
#7  0x00007fffdb8d4373 in baci::CharacteristicComponentImpl::__execute (
    this=0x7fffb4003bd0) at baciCharacteristicComponentImpl.cpp:183
#8  0x00007ffff7b7ee95 in maci::ContainerImpl::activate_component (
    this=0x63bb58, h=16777217, execution_id=1581590620976, 
    name=0x7fffa4003c30 "MANAGEMENT/Gavino", 
    exe=0x7fffa4003e50 "SchedulerImpl", 
    type=0x7fffa4003e20 "IDL:alma/Management/Scheduler:1.0")
    at maciContainerImpl.cpp:2288
---Type <return> to continue, or q <return> to quit---q

closed 2020-02-18 by aorlati