discos/discos #518: KBand receivers cryo temperature read wrongly with connection problems

jjpopski, 2020-02-18:
On Sat 15 Feb 2020  KBand Receivers gave a KBand receiver connection error. Reading the cryo temperatures was still possible; objexp returned values as the cryo system was correctly working (< 15 K LNA temp). However, on Monday, the receiver was found to have temperature about 300 K.

aorlati, 2020-02-18:
I think that the only workaround is to return a placeholder value (negative ?) when the connection is not up and running. That should be easy to implement.

giuseppe-carboni, 2020-02-18:
That was my guess too, it is likely that the component returns the last read values before the connection was lost. We can either:
1 - ask for new valules to the hardware in real time
or
2 - return last read values only if they were taken not too long ago
For the latter solution, the class StatusUpdater acts as a watching thread, it should be enough to just add some checks for the timestamp of the last successful connection.

aorlati, 2020-02-19:
Solution 2 is exactly what is being done right now. I think we can just add a check of the connection. If it is ok we can return the last available readout otherwise we should return a value (a zero or e negative) that highlights the problem. Also we can return the timestamp (from the read method of the devio) the time of the last readout.
This change should be propagated in all receivers component (also including Noto and Medicina) in order to maintain consistency.

marco-buttu, 2020-02-25:
I ran the following test using the `7GHz`:

* get a property value
* cut off the board connection by stopping the simulator
* get a property value
* make the connection available by restarting the simulator

The test confirms the wrong behavior. In fact we are catching the communication error but not setting the error completion and a dummy value for the property (like `-99999`). We can fix the issue by setting both fields in the catch clause. Unfortunately there is another problem... When the board becomes available, the component does not make a fresh connection, so we have to close the component and restart it again. I propose to separate the two misbehaviours, hence the solutions: 

1. Return a dummy value `-99999` in case of communication error, and set the error completion
2. Make a new connection to the board in case you got a communication error

Implementing point `1` we still have a partial solution, but actually it will be an effective one: in case of communication error the temperature will be `-99999`, it means we have to restart the component. After restarting, if we are still getting `-99999` it means there is no communication to the board, otherwise it means we are reading the proper temperature value. In both cases we need to check the board (because we had this problem once in 10 years, so in all likelihood there is something wrong either with the board or its wirings).

aorlati, 2020-02-25:
I agree to implement point 1 which should be simple and address 99% of the cases. We have to check the error (inside the ComponetCore class, the flag should already be there). If an error is detected we have two possible ways, return a dummy value (and the timestamp of the last successful readout) or raise an error flag. In the latter case read devio::read should raise an exception, the completion will be filled accordingly in the get_sync method of the property.

aorlati, 2020-02-25:
Implementing point 2 would be nice, but it requires a change in the ReceiversLibrary.

giuseppe-carboni, 2020-02-25:
I wonder if it's really necessary to keep the connection open all the time. From my understanding, the connection is needed whenever a command is issued to the receiver, or when one or more parameters has to be read. Could we consider switching to this strategy?

marco-buttu, 2020-02-25:
Yes, it is maybe the easiest way to fix point 2. I opened another issue (#522), we can discuss there about point 2.

marco-buttu, 2020-02-25:
I am taking care of this issue, opening a new branck (fix-issue-518)

marco-buttu, 2020-02-28:
Fixed in commit ea1621a.

aorlati, 2020-08-24:
https://github.com/discos/discos/pull/523#issuecomment-679126912

closed 2020-09-04 by aorlati