discos/discos #522: Receivers components do not reconnect to the board

marco-buttu, 2020-02-25:
Once a receiver component loose the connection to the board, it does not make a new fresh connection, that is why the component is not able anymore to get the current property value, also if the board is online. We currently have to restart the receiver component. Related to issue #518.

giuseppe-carboni, 2020-02-25:
As mentioned in #518:
I wonder if it's really necessary to keep the connection open all the time. From my understanding, the connection is needed whenever a command is issued to the receiver, or when one or more parameters has to be read. Could we consider switching to this strategy?

marco-buttu, 2020-02-25:
Yes, to me the easiest and straightforward way to fix the problem is the one you suggest.

giuseppe-carboni, 2020-02-25:
This will allow other clients to connect to the receivers as well even if the component is active, provided that the other client starts the connection when the component is not occupying the socket. In this case the re-connection attempts from the component should notify the failure in opening the socket again.

marco-buttu, 2020-02-25:
Mmm, I think the fact that a client could get the connection preventing DISCOS to communicate to the receivers is a downside of this strategy...

aorlati, 2020-02-26:
The risk is the monitoring software (used by RF people) takes control of the boards keeping Discos out. If this must not happen during an observation.