discos/discos #524: Some LPComponent misbehaviour

marco-buttu, 2020-02-28:
There is an issue with the LP component. Here are the steps required to reproduce it:

* Suricate is running and monitoring the LP component
* The LP component is running
* The LP component is active on `objexp`
* Using `objexp` ask for a LP property value (not sure it is required to spot the issue)
* Close the LP container
* Using `objexp` (not a new instance but the previous objexp instance) ask for a LP property value: it will raise a `CORBA.TRANSIENT` exception
* Restart the LP container: **it will crash after a few seconds!**

I tryed to reproduce the issue with the C and K band components, but they work fine. I have two questions to answer:

1. Why does it only happen when suricate is active?
2. What is wrong with the LP component?

If I fix the problem by changing something in Suricate (maybe the component release or something else...), I also have to come back to the old Suricate versione (the one that spots the LP problem) in order to understand and fix the LP component problem, also if it does not appear anymore after changing Suricate.

marco-buttu, 2020-03-11:
I finally found the source of the problem: the `vd` (`vdL1`, `vdR1`, ..., `vdL3`, `vdR3`), `id` and `vg` properties. Suricate is monitoring them, that is why the issue only happens when Suricate is running. But you can easily spot the problem without Suricate, by following these steps:

1. Run ACS, `SRTLPBandContainer` and `LocalOscillatorLPContainer`
2. **DO NOT** activate the `SRTLPBandReceiver`
3. The following script breaks the container:

```python
from __future__ import print_function
from Acspy.Clients.SimpleClient import PySimpleClient

try:
    client = PySimpleClient()
    lp = client.getComponent('RECEIVERS/SRTLPBandReceiver')
    property_obj = lp._get_idL1()
    value, comp = property_obj.get_sync()  # Line X
except Exception, ex: 
    print('ERROR: %s' % str(ex))
else:
    print('Everything OK...')
finally:
    client.disconnect()
```

When the script executes *Line X*, the container crashes. I think because there is something not yet activated in order to get that property (and also `vd` and `vg` properties). In fact, you can *activate* the component by following the previous two steps, **commenting all Line X*  and running the script once. After that you can decomment *Line X* and run the script again: now you are able to get the property value. If the container crashes again, than you have to *activate* it before asking again for the property.

Question: why can you get the property by `objexp` whitout any previous *activation*? What is this kind of missing *activation*? Probabily `objexp' performs it when you ask for a property...

aorlati, 2020-03-11:
indeed, weired bug! In principle the properties are created when the component is. So you should not be able to activate the component itself......
Does 'gdb' say something more? Does this happen also with all other properties?

marco-buttu, 2020-03-11:
Yes weird, looking for it since yesterday :-D It only happens with the LNA currents and voltages properties (`vdXX`, `idXX`, `vgXX`). I did not investigate with gbd because I was working with screen, not able to scroll up all the output (so I only got the line). By the way, is there a way to save to file the gbd output?

aorlati, 2020-03-11:
are all these properties the only sequence properties? 
Some ACS bug in the implementation of that type of property?

marco-buttu, 2020-03-12:
The `vd`, `id` and `vg` are `ROdobuleSeq`, and we have a lot of them in the receivers, but the issue is only with `vd`, `id` and `vg`.

aorlati, 2022-05-27:
@marco-buttu I know it's a long time since we started this discussion, but I wonder if we can say more on this issue. Can we close it or not? Do you need to test it again?

marco-buttu, 2022-05-27:
@aorlati yes you can close it. In case we decide to investigate, we open it again

closed 2022-05-27 by aorlati