discos/discos #532: MinorServoContainer CPU percentage

marco-buttu, 2020-03-25:
Using `top` to look at the resources on `discos-manager` I see the `MinorServoContainer` is taking more than 100% CPU. It is because of the too strict sleep time of some threads:

```
const unsigned long scheduler_sleep_time = 20 * 10;  // 20us
const unsigned long listener_sleep_time = 2 * 10;  // 2us
const unsigned long status_sleep_time = 2000 * 10; // 2ms
```
I suggest to change these parameters as follow:

```
const unsigned long scheduler_sleep_time = 30000 * 10;  // 30ms
const unsigned long listener_sleep_time = 10000 * 10;  // 10ms
const unsigned long status_sleep_time = 80000 * 10; // 80ms
```

aorlati, 2020-03-25:
This is a duplicate of the issue #169. Please close them both when this is addressed.

marco-buttu, 2020-03-25:
Fixed in commit 1670e96. Also closed the duplicate (issue #169)

aorlati, 2022-03-16:
closing this issue since it's no more relevant as the minor servo system control of the SRT will be replaced and so discos component will be completely reworked.  Check #647