discos/discos #535: ManagementContainer breaks operatorInput

marco-buttu, 2020-03-31:
Have a look at this scenario:

```python
>>> client = PySimpleClient()
>>> component = client.getComponent('RECEIVERS/SRTKBandMFReceiver')
```
At this point you stop the `SRTKBandContainer` and restart it again. The client is no more logged in:

```python
>>> component.activate('KKG')
Traceback (most recent call last):
   ...
COMM_FAILURE: CORBA.COMM_FAILURE(...)
```

This only happens with CPP components/containers. If you restart a Python container, the client attached to its components still work. 

Here is the point of this issue: the `operatorInput` gets a client during startup, and its component (the scheduler) is implemented in CPP. It means that you can not use anymore that client after stopping the scheduler container, no matter if you restart the container. In that case you need to restart the `operatorInput` as well (after restarting the container).

We can fix this issue in several ways. The esiest one is to get a fresh client and component for every command, and release component/client after execution. A second one is to manage the exception and get a new client in case of CORBA failure. **Be careful when managing CORBA exceptions**. Look at this:

```python
>>> component.activate('KKG')
Traceback (most recent call last):
   ...
COMM_FAILURE: CORBA.COMM_FAILURE(...)
>>> component.activate('KKG')
Traceback (most recent call last):
    ...
OBJECT_NOT_EXIST: CORBA.OBJECT_NOT_EXIST(...)
```

The second call raises a different exception.

aorlati, 2022-05-27:
@marco-buttu I do not remember if this is ever come to a conclusion. In case I think that a good trade of is to try to intercept the CORBA.COMM_FAILURE exception and i that case try to renew the component reference (.getComponent).

marco-buttu, 2022-05-27:
@aorlati I did not fix the issue because a I left to Antartica. We keep it open, I am going to fix it as soon as I can (late summer or fall)