discos/discos #539: 'project' command on the operatorInput could return a message to the user

giuseppe-carboni, 2020-05-04:
It could be a good idea to print to the user a message containing a link/useful information about the state of the system/telescope.

aorlati, 2020-05-14:
expanded parser in order to support string as return value. 
Now we should code the project command so that a well come message is  retrieved from CDB.
An had hoc DB entry should be created in DataBlock/.
Also I've set the checkProjectCode in Scheduler configuration to true. I can't see any reason to keep it false.

aorlati, 2020-05-21:
@giuseppe-carboni did you check the branch? Are you going to complete the coding required to close the issue.

giuseppe-carboni, 2020-05-21:
Yes, I'm working on it and I will push some code soon

commit eaa728a9 (2020-05-21): Fix #539. Added a custom welcome message for the SRT station.

The message reminds the user, whenever the `project` command is issued, to check the status of the telescope in the relative webpage.
commit 80d7e414 (2020-05-26): Fix #539, moved the welcomeMessage parameter to DataBlock
commit 2a3c4d28 (2020-05-26): Fix #539, parser now supports strings as return. Added a welcome message to the user (SRT). (#543)

* Fix issue #539: parser expanded in order to support strings as return value. logString_type added to types list.
This now should allow to 'project' command to send a wellcome message to the user.

* Fix #539. Added a custom welcome message for the SRT station.

The message reminds the user, whenever the `project` command is issued, to check the status of the telescope in the relative webpage.

* Fix #539, moved the welcomeMessage parameter to DataBlock
commit 204e91b7 (2022-02-09): Fix #539, parser now supports strings as return. Added a welcome message to the user (SRT). (#543)

* Fix issue #539: parser expanded in order to support strings as return value. logString_type added to types list.
This now should allow to 'project' command to send a wellcome message to the user.

* Fix #539. Added a custom welcome message for the SRT station.

The message reminds the user, whenever the `project` command is issued, to check the status of the telescope in the relative webpage.

* Fix #539, moved the welcomeMessage parameter to DataBlock
closed 2020-05-26 by giuseppe-carboni