discos/discos #556: Mail delivery support API

aorlati, 2020-09-28:
A support for sending email to system administrators should be added. 
This has implications also to system deployment ad the mail service should be configured.

This call should be added to Noto and Medicina mount control when a pendolation of the servo system is detected

aorlati, 2020-09-29:
In order to make this enhancement work a proper configuration of the mail service should be done in the workstation. 
This will involve the deployment system and an issue to be open (@giuseppe-carboni).
Particularly 
1) install agent "postfix"
2) configure it with a proper mail server. The file do be set is "/etc/postfix/main.cf" changind the following keyword accordingly
`myhostname =` 
`relayhost = `

giuseppe-carboni, 2020-10-20:
Seems like no further configuration is needed when the mail is sent by a machine under a station's domain. Same applies with virtual machines. When the machines is in SRT or Medicina network, the mail gets sent successfully. I cannot send any mail with the `mail` command when working from home.

giuseppe-carboni, 2020-11-13:
We forgot to add `DiscosLocals` library in the main Makefile

commit 2fb5d74d (2020-11-13): Fix #556, updated main Makefile
commit 8a2348a2 (2020-11-16): Fix #556 again, added `DiscosLocals` in main Makefile (#567)

* fix issue #556: Implemented API to send eMail to system administrators. This is now used when a pendolation in Noto and Medicina servo system is
detected. Also moved the welcome message from station.xml configuration file to StationLocals.xml.

* Fix #556, updated main Makefile
commit 6b90e8a1 (2022-02-09): Fix #556, updated main Makefile
closed 2020-11-16 by giuseppe-carboni