discos/discos #585: Schedule with no backend and no recorder not properly handled.

aorlati, 2021-05-10:
In other word the scan definition as reported below was not properly handled:
SC: 1  label NULL:NULL
The scheduler does not wait for the time assigned to the subscan and NULL is simply interpreted as no recording needed so the next subscan is immediately started.
Moreover the NULL recorder generates and error ad the scheduler try to instantiated it even if there is no need to record.

giuseppe-carboni, 2021-05-13:
Issue fixed in #586, closing it.

aorlati, 2021-07-08:
This fix has an hidden bug. The scheduler now does not handle properly file creation (names and folders) so that all files of a schedule are created inside the same folder.  I'm reopening this issue.