discos/discos #587: ManagementContainer, "Error in schedule reporting" after haltSchedule/stopSchedule

giuseppe-carboni, 2021-05-17:
This warning message should not appear since it was the user to choose to halt/stop the schedule

giuseppe-carboni, 2021-05-31:
@aorlati I believe the issue is here:
https://github.com/discos/discos/blob/master/Common/Servers/Scheduler/src/ScheduleExecutor.cpp#L793
which calls the following method:
https://github.com/discos/discos/blob/master/Common/Servers/Scheduler/src/ScheduleReport.cpp#L110
which returns false for an error somewhere, or true when no error is found.
I believe the condition in the first if should be inverted by heading the call with a `!`.
Asking your opinion on this since I'm not 100% sure.

aorlati, 2021-05-31:
Hit the head of the nail! 
You can fix it and issue a PR.
Thanks.

commit 75ecda1c (2021-06-07): Fix #587, fixed scheduler reporting false error (#591)
closed 2021-06-07 by giuseppe-carboni