discos/discos #604: Check consistency between the currently active project and the project name stated in the schedule header

aorlati, 2021-09-22:
If schedule keyword PROJECT differs from the currently active project (command `project=code`)  the data archiving system runs into consistency problem that requires manual fixes. In order to avoid this DISCOS should make sure the observer is running a schedule that belongs to its project.

aorlati, 2021-09-22:
The keyword PROJECT is mandatory in the schedule header, so a good approach is to check that it matches with the currently active project. In case  there is no correspondence the execution is aborted and the user is presented with a warning message that ask to check the schedule keyword. 
In order to clarify the currently active project can be declared by the user directly with the 'project' command or  indirectly with the 'startSchedule' command ('startSchedule=11-23/schedulename.scd,1').

alle-ira, 2021-09-23:
I agree on aborting the observation if there is no correspondence. It has also the advantage to "train" the observers to write appropriate values in the schedules in advance.
I would prefer that the active project is set via the startSchedule command ('startSchedule=11-23/schedulename.scd,1'). The use of the PROJECT command should be discouraged for normal users, it is prone to typos or in general mistakes/mis-use.

jjpopski, 2021-09-23:
I do agree.

giuseppe-carboni, 2021-11-18:
Seems like starting a schedule with a non-matching project id leads to a crash of the ManagementContainer, this will have to be investigated

aorlati, 2022-01-18:
I confirm this is the case, I'm uploading a fix

closed 2022-01-31 by aorlati