discos/discos #655: Adding a check for currently commanded mode for Noto and Medicina ACU

aorlati, 2022-01-04:
This mode check will be under the control of the CDB (Noto will now be enabled, Medicina disabled). 
The software should check if the commanded mode is currently active, if not it will try to recover. 
This feature is needed to counter measure seldom event when the ACU loose its configuration (power failure ecc....)

fvitello, 2022-03-28:
Merged https://github.com/discos/discos/pull/692/

aorlati, 2022-09-26:
Partially unexpected  behaviour in Noto after a power failure. The event is correctly detected and the error cleared but the servo system fails to reactivate the drive and motor power. The ACU remains in a proper mode leading to no errors but no telescope movement is done.

aorlati, 2022-09-26:
No real evidence of the problem, I'm implementing a couple of tricks that should improve the event handling by DISCOS then leading to a resolution of the problem.