discos/discos #719: In schedule .bck file, the last command is ignored when followed by the } character

giuseppe-carboni, 2022-06-16:
For example, in this file:
`SARDARA:BACKENDS/Sardara{
enable=3;6}`
The enable command is recognized (the ManagementContainer doesn't show any error when configuring recording) but somehow it is ignored due to the } character being on the same line. If the } character is sent to a new line, the enable command is recognized and executed correctly.
The scheduleChecker command also reports no error in the schedule.
@marcolandoni I'm also tagging you for reference on this one since the bug is introduced by basie, I can open an issue there as well

aorlati, 2022-06-17:
I'll have a look at the issue asap.
@marcolandoni, do you think we should post an issue for Basie as well?

marcolandoni, 2022-06-17:
Hi @aorlati ,
Yes! I update the code according to what we said. There was an error in the key definition and a typo due to a manual (mine) change on the schedule

closed 2022-09-14 by aorlati