discos/discos #744: Force the commanded elevation to move theAS for testing

aorlati, 2022-09-14:
AS Boss interface will be added with a method in order to provide an elevation value. The Boss component will move the actuators into the correct position for the provided elevation. This will be used for testing and troubleshooting activities only.

giuseppe-carboni, 2022-09-15:
By using objexp, select the 'By type' tab, AS, Boss and call the following methods:
1) 'setup' with one of the following values: 'S' for shaped, 'P' for parabolic, 'SF' for shaped fixed and 'PF' for parabolic fixed (these are the available profiles for the SRT, Medicina and Noto may vary)
2) 'update' with a floating point number representing the elevation (in degrees).
This should set all the actuators to the position they would move to during tracking at the given elevation.

carlomigoni, 2022-09-15:
Another method other then described by Giuseppe (but for SRT only at the moment) is by using the SRTActiveSurfaceManagementGUIClient where you can find the setup command for choosing the profile and the update command with an entry where put the elevation value.

giuseppe-carboni, 2023-03-28:
> This behavior can already be commanded. By using objexp, select the 'By type' tab, AS, Boss and call the following methods:
> 
>     1. 'setup' with one of the following values: 'S' for shaped, 'P' for parabolic, 'SF' for shaped fixed and 'PF' for parabolic fixed (these are the available profiles for the SRT, Medicina and Noto may vary)
> 
>     2. 'update' with a floating point number representing the elevation (in degrees).
>        This should set all the actuators to the position they would move to during tracking at the given elevation.
> 
> 
> EDIT: fixed profiles will not let you update the USDs positions, you will have to use either Shaped or Parabolic profiles for this.

Accepting this as proposed method for all stations. Closing.

closed 2023-03-28 by giuseppe-carboni