discos/discos #784: Script discosdown does not remove the file lock

marco-buttu, 2022-12-15:
The current version of ACS changed the behaviour of `killACS`, now it does not remove the file lock. In the script `discosdown` we are calling `killACS`, we should change to `acsStop`.

giuseppe-carboni, 2022-12-15:
acsStop takes a very long time to shutdown ACS, I'm currently using `killACS`, which kills the processes rather quickly, followed by `acsStop`, which will only take care of removing the file lock. I suggest we follow this strategy.

aorlati, 2022-12-15:
Can you provide more details? where is the script discosdown? DISCOS or Provisioning? 
The other question: what about the acscommandcenter, did they change acstop e killacs buttons as well?

commit 0cb9a21c (2023-06-21): Fix #784, added `acsStop` to `discosdown` script (#798)
closed 2023-06-21 by giuseppe-carboni