discos/discos #825: Split FITS NODATA directory and final FITS directory

giuseppe-carboni, 2024-01-08:
FITS NODATA are always written in the dataPath directory as shown [here](https://github.com/discos/discos/blob/master/SRT/Configuration/CDB/alma/DataBlock/Equipment/Equipment.xml#L25). My suggestion is to use that directory only for the NODATA files, as a temporary directory. The final directory should not contain any temporary file used by the backend merging algorithm. This is my suggestion:
DISCOS machines should have a directory for the NODATA fits files, which will be exported read only via NFS. Backends should mount that directory, read the NODATA FITS files, perform the merging procedure, and write the final files in the lustre exported filesystem, having directory /discos-archive/data/<projectid>/<date>/. This will allow for a cleaner data directory.
The only modification that I suspect is needed, is to write the FITS NODATA in the said exported directory, and write the summary FITS file in the original /discos-archive/data/etc path.

aorlati, 2024-01-08:
@giuseppe-carboni the nodata files are still important for debugging reasons. They asked to preserve them as part of the dataset. Summary files are written in the relevant path (/discos-archive/data/....) whilst nodata are stored in (same path)/tmp. In this case you do not need to export other NFS drives. Medicina has this configuration, see [configuration file](https://github.com/discos/discos/blob/master/Medicina/Configuration/CDB/alma/DataBlock/Equipment/Equipment.xml)

giuseppe-carboni, 2024-01-09:
I see, I didn't know about the necessity to keep the NODATA alongside the final fits, then we will use the strategy already applied to Medicina. Closing.

closed 2024-01-09 by giuseppe-carboni