discos/discos #883: ManagementContainer segfault when receiver sections do not match backend sections during tsys

giuseppe-carboni, 2024-07-10:
This bug emerged in the following configuration:
SRT station:
receiversSetup=QQG (the receiver has 19 feeds configured in the CDB DataBlock directory)
chooseBackend=TotalPower
initialize=QQG (the backend has been configured with only 2 feeds)
tsys (ManagementContainer segfault)

This is probably due to the fact that a loop inside the tsys calculation function expects 19*2 sections for both the receiver and the backend. I don't know yet how we should handle this but we should definitely avoid having a segmentation fault.

giuseppe-carboni, 2024-07-12:
This might not be the case, I found out that the tsys calculation function was crashing the container even if the number of sections between the receiver and the backend matched. It was probably an error due to the wrong NoiseMark table used.
I will investigate further and act accordingly.

aorlati, 2024-07-12:
@giuseppe-carboni I've just sent some additional information about the NoiseMark table and how it should be created. I suggest to go through that and than check if the issue persists.