discos/discos #895: Summary FITS file: missing parameters

coding-bricks, 2024-12-19:
Dear All,

in order to work efficiently, the quick-look which I am developing  (both off-line and on-line version) needs to get access to some parameters (i.e. for example, "rest frequency" and "radial velocity") which are stored in the summary FITS file. The latter is created once the schedule starts but, however, doesn’t provide the required parameters. This happens only once it gets closed, that is, once the schedule is stopped. As a consequence, the quick-look is not able to produce the requested plots scan after scan but only at the end of the schedule.  
Question: is it possible to provide the requested parameters immediately once the summary FITS file gets created?

Regards,
Fabio

aorlati, 2025-03-14:
The summary file is created at the end of the scan because some information (i.e. total integration time) are know only at the end. In order to change this (incremental creation of the summary file) a large recoding is required. 
To cope with this request I will add the rest frequency keywords (in principle a rest frequency can be specified for each section) in the header of the Section table. I think this data duplication is acceptable because the total size is very limited.

aorlati, 2025-04-06:
I'm closing this issue. The fix has been merged (#914) into development branch.

closed 2025-04-06 by aorlati